tbsdtv/linux_media #374: Usb Tuner TBS5930 dvb s2x with E2 system linux (Dreambox)

fairbird, 2024-12-23:
I'm tring to support for Usb Tuner TBS5930 with E2 devices such as dreambox. 
I have cherry-pick the codes and files from your source and created this patch with some edit to make it compatible with E2 system kernel 3.14.
https://github.com/fairbird/openpli-dreambox-oe-core/commit/eafe53aa0cc2269f434dd32faa9996d0b9ac103e
and
https://github.com/fairbird/openpli-dreambox-oe-core/commit/cdcc40aac168cfa3512fe686645aaba11aed0a20

And also sent drivers (dvb-usb-id5930 + dvb-demod-m88rs6060.fw) to (/lib/firmware)

But the tuner does not appear on tuner screen option ...
And here the dmesg result !
[dmesg.txt](https://github.com/user-attachments/files/18229067/dmesg.txt)

crazycat69, 2024-12-23:
modules must me in /lib/modules/

fairbird, 2024-12-23:
> root@dm900:~# modprobe dvb-usb-tbs5930
modprobe: FATAL: Module dvb-usb-tbs5930 not found in directory /lib/modules/3.14-1.17-dm900

crazycat69, 2024-12-23:
so 5930 driver not installed

fairbird, 2024-12-23:
No need drive I think as like (TBS5980 TBS5928 TBS5925 TBS5921 TBS5920 TBS5910)
I think need to update this patch
https://github.com/fairbird/openpli-dreambox-oe-core/blob/develop/meta-dreambox/recipes-linux/linux-dreambox-3.14/0001-Support-TBS-USB-drivers.patch
 but the information I do not know exactly such as this code what is correct code huld I have add for 5930 is it like 5980 or 5928 or 5910 ?!!
https://github.com/fairbird/openpli-dreambox-oe-core/blob/develop/meta-dreambox/recipes-linux/linux-dreambox-3.14/0001-Support-TBS-USB-drivers.patch#L630-L765

crazycat69, 2024-12-23:
Look like 5930, m88rs6060 modules not compiled.
Need add to defconfig for each STB (DM900, 920)
https://github.com/fairbird/openpli-dreambox-oe-core/blob/develop/meta-dreambox/recipes-linux/linux-dreambox-3.14/dm900/defconfig
https://github.com/fairbird/openpli-dreambox-oe-core/blob/develop/meta-dreambox/recipes-linux/linux-dreambox-3.14/dm920/defconfig

fairbird, 2024-12-23:
I have it already 
https://github.com/fairbird/openpli-dreambox-oe-core/blob/develop/meta-dreambox/recipes-linux/linux-dreambox-3.14/dm900/defconfig#L2978

But I think as I wrote we need to update this patch with correct codes
https://github.com/fairbird/openpli-dreambox-oe-core/blob/develop/meta-dreambox/recipes-linux/linux-dreambox-3.14/0001-Support-TBS-USB-drivers.patch

And we have only these drivers for TBS-USB but what driver should be work with 5930. So I can update the patch depend on it correct driver. This is the point.
> root@dm900:# ls -n /lib/modules/3.14-1.17-dm920/kernel/drivers/media/dvb-frontends
-rw-r--r--    1 0        0            21436 Mar 31  2014 cx24116.ko
-rw-r--r--    1 0        0             5844 Mar 31  2014 stb6000.ko
-rw-r--r--    1 0        0             9096 Mar 31  2014 stb6100.ko
-rw-r--r--    1 0        0            11248 Mar 31  2014 stv0288.ko
-rw-r--r--    1 0        0            64352 Mar 31  2014 stv090x.ko
-rw-r--r--    1 0        0            12856 Mar 31  2014 tda10071.ko

crazycat69, 2024-12-24:
ok **CONFIG_DVB_M88RS6060** present, but need **CONFIG_DVB_USB_TBS5930** also :)

fairbird, 2024-12-24:
Ok .. I will test also CONFIG_DVB_USB_TBS5930 !

fairbird, 2024-12-25:
@crazycat69 
After Adding (CONFIG_DVB_M88RS6060 and CONFIG_DVB_USB_TBS5930) in  defconfig and compiled driver ...
https://github.com/fairbird/openpli-dreambox-oe-core/blob/develop/meta-dreambox/recipes-linux/linux-dreambox-3.14/dm900/defconfig#L2979
https://github.com/fairbird/openpli-dreambox-oe-core/blob/develop/meta-dreambox/recipes-linux/linux-dreambox-3.14/dm900/defconfig#L2610
The deives (Dm900) does not boot (freezing in boot) and after remove usb tuner the device boot again normally .
So the driver of dvb-usb-tbs5930 does not work probable!
The patch I think not correctly !
https://github.com/fairbird/openpli-dreambox-oe-core/blob/develop/meta-dreambox/recipes-linux/linux-dreambox-3.14/0001-add-support-for-tbs5930.patch

fairbird, 2024-12-26:
@crazycat69

Here is new patch based on your files from this source without change any codes
[0001-add-support-for-tbs5930.patch.txt](https://github.com/user-attachments/files/18250962/0001-add-support-for-tbs5930.patch.txt)

And here is what I have got from errors ! And advice ?!

> /home/raed/build_image/My-Openpli/openpli-dreambox-oe-core/build/tmp/work-shared/dm900/kernel-source/drivers/media/usb/dvb-usb/tbs5930.c:171:9: error: implicit declaration of function 'strscpy'; did you mean 'strlcpy'? [-Werror=implicit-function-declaration]
  171 |         strscpy(info.type, "m88rs6060", I2C_NAME_SIZE);
      |         ^~~~~~~
      |         strlcpy
/home/raed/build_image/My-Openpli/openpli-dreambox-oe-core/build/tmp/work-shared/dm900/kernel-source/drivers/media/usb/dvb-usb/tbs5930.c:175:18: error: implicit declaration of function 'i2c_new_client_device'; did you mean 'i2c_new_probed_device'? [-Werror=implicit-function-declaration]
  175 |         client = i2c_new_client_device(&d->i2c_adap,&info);
      |                  ^~~~~~~~~~~~~~~~~~~~~
      |                  i2c_new_probed_device
/home/raed/build_image/My-Openpli/openpli-dreambox-oe-core/build/tmp/work-shared/dm900/kernel-source/drivers/media/usb/dvb-usb/tbs5930.c:175:16: warning: assignment to 'struct i2c_client *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
  175 |         client = i2c_new_client_device(&d->i2c_adap,&info);
      |                ^
  LD [M]  drivers/media/usb/dvb-usb-v2/dvb-usb-mxl111sf.o
/home/raed/build_image/My-Openpli/openpli-dreambox-oe-core/build/tmp/work-shared/dm900/kernel-source/drivers/media/usb/dvb-usb/tbs5930.c:176:13: error: implicit declaration of function 'i2c_client_has_driver' [-Werror=implicit-function-declaration]
  176 |         if(!i2c_client_has_driver(client))
      |             ^~~~~~~~~~~~~~~~~~~~~
  CC [M]  net/netfilter/xt_esp.o
/home/raed/build_image/My-Openpli/openpli-dreambox-oe-core/build/tmp/work-shared/dm900/kernel-source/drivers/media/dvb-frontends/m88rs6060.c:1932:42: error: array type has incomplete element type 'struct reg_sequence'
 1932 |         static const struct reg_sequence reset_buf[] = {
      |                                          ^~~~~~~~~
  CC [M]  drivers/net/wireless/ti/wlcore/main.o
/home/raed/build_image/My-Openpli/openpli-dreambox-oe-core/build/tmp/work-shared/dm900/kernel-source/drivers/media/dvb-frontends/m88rs6060.c:2129:14: error: 'struct dtv_frontend_properties' has no member named 'scrambling_sequence_index'
 2129 |         if (c->scrambling_sequence_index) {
      |              ^~
/home/raed/build_image/My-Openpli/openpli-dreambox-oe-core/build/tmp/work-shared/dm900/kernel-source/drivers/media/dvb-frontends/m88rs6060.c:2131:29: error: 'struct dtv_frontend_properties' has no member named 'scrambling_sequence_index'
 2131 |                 pls_code = c->scrambling_sequence_index;
      |                             ^~
/home/raed/build_image/My-Openpli/openpli-dreambox-oe-core/build/tmp/work-shared/dm900/kernel-source/drivers/media/dvb-frontends/m88rs6060.c:1932:42: warning: unused variable 'reset_buf' [-Wunused-variable]
 1932 |         static const struct reg_sequence reset_buf[] = {
      | 
/home/raed/build_image/My-Openpli/openpli-dreambox-oe-core/build/tmp/work-shared/dm900/kernel-source/drivers/media/dvb-frontends/m88rs6060.c:3201:33: error: 'APSK_64' undeclared (first use in this function); did you mean 'APSK_32'?
 3201 |                 p->modulation = APSK_64; break;
      |                                 ^~~~~~~
      |                                 APSK_32
/home/raed/build_image/My-Openpli/openpli-dreambox-oe-core/build/tmp/work-shared/dm900/kernel-source/drivers/media/dvb-frontends/m88rs6060.c:3201:33: note: each undeclared identifier is reported only once for each function it appears in
/home/raed/build_image/My-Openpli/openpli-dreambox-oe-core/build/tmp/work-shared/dm900/kernel-source/drivers/media/dvb-frontends/m88rs6060.c:3207:72: error: 'FEC_1_4' undeclared (first use in this function); did you mean 'FEC_3_4'?
 3207 |                         case MtFeCodeRate_1_4:          p->fec_inner = FEC_1_4;         break;
      |                                                                        ^~~~~~~
      |                                                                        FEC_3_4
/home/raed/build_image/My-Openpli/openpli-dreambox-oe-core/build/tmp/work-shared/dm900/kernel-source/drivers/media/dvb-frontends/m88rs6060.c:3208:72: error: 'FEC_1_3' undeclared (first use in this function); did you mean 'FEC_2_3'?
 3208 |                         case MtFeCodeRate_1_3:          p->fec_inner = FEC_1_3; break;
      |                                                                        ^~~~~~~
      |                                                                        FEC_2_3
/home/raed/build_image/My-Openpli/openpli-dreambox-oe-core/build/tmp/work-shared/dm900/kernel-source/drivers/media/dvb-frontends/m88rs6060.c:3218:72: error: 'FEC_5_9' undeclared (first use in this function); did you mean 'FEC_8_9'?
 3218 |                         case MtFeCodeRate_5_9:          p->fec_inner = FEC_5_9; break;
      |                                                                        ^~~~~~~
      |                                                                        FEC_8_9
[... cut ...]
/home/raed/build_image/My-Openpli/openpli-dreambox-oe-core/build/tmp/work-shared/dm900/kernel-source/drivers/media/dvb-frontends/m88rs6060.c:3346:24: note: (near initialization for 'm88rs6060_ops.set_tone')
/home/raed/build_image/My-Openpli/openpli-dreambox-oe-core/build/tmp/work-shared/dm900/kernel-source/drivers/media/dvb-frontends/m88rs6060.c:3347:10: error: 'const struct dvb_frontend_ops' has no member named 'eeprom_write'
 3347 |         .eeprom_write = m88rs6060_eeprom_write,
      |          ^~~~~~~~~~~~

crazycat69, 2024-12-26:
Need backport this drivers to 3.14 kernel

fairbird, 2024-12-26:
How. Any patches ?!

fairbird, 2024-12-31:
@crazycat69
Finally the tuner appear but the problem is (Empty) >>> Is it from drivers ?!

[image]

crazycat69, 2024-12-31:
show kernel log

fairbird, 2025-01-01:
Here the log of dmesg
[dmesg.txt](https://github.com/user-attachments/files/18286955/dmesg.txt)

And do you think I need to add TSB-5930 code here ?!
https://github.com/fairbird/enigma2-dreambox/blob/develop/lib/dvb/frontend.cpp#L1282
if yes what you suggestion about this code ?!
https://github.com/fairbird/enigma2-dreambox/blob/develop/lib/dvb/frontend.cpp#L1283-L1286

crazycat69, 2025-01-02:
Look like some wrong with frontend name reading in eDVBUsbAdapter::eDVBUsbAdapter()
https://github.com/fairbird/enigma2-dreambox/blob/develop/lib/dvb/dvb.cpp#L305