tbsdtv/linux_media #375: build error on Ubuntu 24.04.1 LTS (kernel 6.8.0-51)

chikatana, 2024-12-31:
I've been using TBS6514 for a while on proxmox VE with PCI-e passthough to an Ubuntu 24.04.1 LTS (kernel 6.8.0-51) VM.
The driver was built in Nov this year on exactly the save setup and it worked fine all along.
I got to reboot my system for some reason yesterday but after a reboot, the TBS6514 card stopped functioning.

I did a lspci showing the card is still detected:
01.00.0 Multimedia controller: TBS Technologies DVB Tuner PCIe Card

I deleted the old driver and do a rebuild and two errors show up.:

/home/k/media_build/v4l/ccs-core.c: In function 'ccs_set_ctrl':
/home/k/media_build/v4l/ccs-core.c:668:21: error: too many arguments to function 'pm_runtime_get_if_active'
  668 |         pm_status = pm_runtime_get_if_active(&client->dev, true);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/k/media_build/v4l/ccs-core.c:22:
./include/linux/pm_runtime.h:75:12: note: declared here
   75 | extern int pm_runtime_get_if_active(struct device *dev);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~
make[4]: *** [scripts/Makefile.build:243: /home/k/media_build/v4l/ccs-core.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/linux-headers-6.8.0-51-generic/Makefile:1925: /home/k/media_build/v4l] Error 2
make[2]: *** [Makefile:240: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.8.0-51-generic'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/k/media_build/v4l'
make: *** [Makefile:26: all] Error 2
k@K-Ubuntu:~/media_build$ make -j4
make -C /home/k/media_build/v4l 
make[1]: Entering directory '/home/k/media_build/v4l'
make -C firmware prep
creating symbolic links...
make[2]: Entering directory '/home/k/media_build/v4l/firmware'
make[2]: Nothing to be done for 'prep'.
make[2]: Leaving directory '/home/k/media_build/v4l/firmware'
make -C firmware
make[2]: Entering directory '/home/k/media_build/v4l/firmware'
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/home/k/media_build/v4l/firmware'
Kernel build directory is /lib/modules/6.8.0-51-generic/build
make -C ../linux apply_patches
make[2]: Entering directory '/home/k/media_build/linux'
Syncing with dir ../../media
Patches for 6.8.0-51-generic already applied.
make[2]: Leaving directory '/home/k/media_build/linux'
make -C /lib/modules/6.8.0-51-generic/build M=/home/k/media_build/v4l  modules
make[2]: Entering directory '/usr/src/linux-headers-6.8.0-51-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  You are using:           gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  CC [M]  /home/k/media_build/v4l/ccs-core.o
  CC [M]  /home/k/media_build/v4l/ccs-pll.o
  CC [M]  /home/k/media_build/v4l/cs3308.o
  CC [M]  /home/k/media_build/v4l/cs5345.o
/home/k/media_build/v4l/ccs-core.c: In function 'ccs_set_ctrl':
/home/k/media_build/v4l/ccs-core.c:668:21: error: too many arguments to function 'pm_runtime_get_if_active'
  668 |         pm_status = pm_runtime_get_if_active(&client->dev, true);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/k/media_build/v4l/ccs-core.c:22:
./include/linux/pm_runtime.h:75:12: note: declared here
   75 | extern int pm_runtime_get_if_active(struct device *dev);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~
make[4]: *** [scripts/Makefile.build:243: /home/k/media_build/v4l/ccs-core.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/linux-headers-6.8.0-51-generic/Makefile:1925: /home/k/media_build/v4l] Error 2
make[2]: *** [Makefile:240: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.8.0-51-generic'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/k/media_build/v4l'
make: *** [Makefile:26: all] Error 2

I created a new VM and try the build process but it gives the same error.
I was stuck and so desperate that I reinstalled Proxmox VE and the Ubuntu VM with every single setting the same as before. But the same error keep occurring.

Is the driver updated and it no longer works on my system?

crazycat69, 2024-12-31:
https://github.com/tbsdtv/linux_media/issues/370

chikatana, 2024-12-31:
I run every command in the sh file but it gives the same error:

#!/bin/bash

# Enable some staging drivers
make stagingconfig

# Disable RC/IR support
sed -i -r 's/(^CONFIG.*_RC.*=)./\1n/g' v4l/.config
sed -i -r 's/(^CONFIG.*_IR.*=)./\1n/g' v4l/.config

echo "V4L drivers building..."
make -j$(nproc)

echo "V4L drivers installing..."
sudo rm -r -f /lib/modules/$(uname -r)/updates/extra

sudo make install

echo "V4L drivers installation done"
echo "You need to reboot..."