tbsdtv/linux_media #376: Virtual machine shuts down after installing the 5520se driver

manb1365, 2025-01-02:
Hello
I have a 5520se.
I have a server that has 6.7 esxi and I want to use tvheadend on it.
I tried my best to do the installation but the virtual machine would shut down immediately after booting. (Both in Ubuntu 24.04 and 20.04)
I did all the things I did in esxi this time in vmware and surprisingly the card was recognized.
I uploaded the vmware machine to esxi and still the same problem and the machine shuts down
Is there a solution?
Virtual machine shuts down after installing the 5520se driver

https://github.com/user-attachments/assets/90c86599-3c18-4e9b-a89b-b303c4ca26e5

crazycat69, 2025-01-02:
VM is evil :)

manb1365, 2025-01-03:
Dear crazycat
This evil has been destroying me for a month. I have done everything I can think of, but surprisingly, there is no problem on vmware, but as soon as I move to esxi 6.7, this problem occurs
Is there a solution to fix this problem?

onehub278, 2025-07-26:
Same here on ESXi 8.0 with a TBS 5220 USB tuner.

When passing the device through to a Linux VM (running TVHeadend), ESXi triggers a crash dump and forcefully powers off the VM immediately after the firmware is loaded. 
There's no graceful shutdown or kernel panic visible from the guest — the VM process (vmx) crashes on the host side.

I exported my tvheadend VM and imported it into VMware Workstation Pro 17 on a Windows 11 laptop. In that environment, the TBS 5220 works with TVHeadend.

Host: VMware ESXi 8.0.1, Build 21987905  
Guest: Ubuntu 22.04.4 LTS, Kernel 5.15.0-86  
Device: TBS 5220
TV software: TVHeadend 4.3-2245  

Jul 22 13:12:05 tvh kernel: [  204.183185] usb 1-1: new high-speed USB device number 3 using ehci-pci
Jul 22 13:12:06 tvh kernel: [  204.741296] usb 1-1: New USB device found, idVendor=734c, idProduct=5220, bcdDevice= 0.01
Jul 22 13:12:06 tvh kernel: [  204.741302] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jul 22 13:12:06 tvh kernel: [  204.741304] usb 1-1: Product: TBS 5220
Jul 22 13:12:06 tvh kernel: [  204.741306] usb 1-1: Manufacturer: TBS-Tech
Jul 22 13:12:06 tvh kernel: [  204.762808] dvb-usb: found a 'TurboSight TBS 5220 DVB-T/T2/C' in cold state, will try to load a firmware
Jul 22 13:12:06 tvh kernel: [  204.762813] dvb-usb: downloading firmware from file 'dvb-usb-tbsqbox-id5220.fw'
Jul 22 13:12:06 tvh kernel: [  204.762814] tbs5220: start downloading TBS5220 firmware
Jul 22 13:12:06 tvh kernel: [  205.027134] dvb-usb: found a 'TurboSight TBS 5220 DVB-T/T2/C' in warm state.
Jul 22 13:12:06 tvh kernel: [  205.027260] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
Jul 22 13:12:06 tvh kernel: [  205.027409] dvbdev: DVB: registering new adapter (TurboSight TBS 5220 DVB-T/T2/C)
Jul 22 13:12:07 tvh kernel: [  205.819324] dvb-usb: MAC address: ff:ff:ff:ff:ff:ff
Jul 22 13:12:07 tvh kernel: [  205.840182] i2c i2c-0: Added multiplexed i2c bus 1
Jul 22 13:12:07 tvh kernel: [  205.840187] si2168 0-0064: Silicon Labs Si2168-A20 successfully identified
Jul 22 13:12:07 tvh kernel: [  205.840189] si2168 0-0064: firmware version: A 2.0.3
Jul 22 13:12:07 tvh kernel: [  205.942910] si2157 1-0060: Silicon Labs Si2157 successfully attached
Jul 22 13:12:07 tvh kernel: [  205.968279] usb 1-1: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5220 DVB-T/T2/C)...