tbsdtv/linux_media #377: TBS6314R no linux driver?

spacie2136, 2025-01-06:
Is TBS6314R still not supported under linux? I do not get any /dev/videoX interfaces but i can see the card listed in lspci like a TBS-DVB card.

tbslucy, 2025-01-06:
Hi,

please have a try the attached driver.
 
cd tbscapture

#make

cd release

#install.sh

then check the video device 

best regards,
lucy

Lucy Lu | Software Engineer
***@***.*** | www.tbsdtv.com
   TBS Technology
THE BEST SOLUTIONS FOR DIGITAL TV AND MORE
Live TV Streaming | IPTV Head-end | OEM & ODM Solution
---------------------------------------------------------------------------------

 
From: spacie2136
Date: 2025-01-06 17:26
To: tbsdtv/linux_media
CC: Subscribed
Subject: [tbsdtv/linux_media] TBS6314R no linux driver? (Issue #377)
Is TBS6314R still not supported under linux? I do not get any /dev/videoX interfaces but i can see the card listed in lspci like a TBS-DVB card.

crazycat69, 2025-01-17:
https://github.com/tbsdtv/tbscapture