tbsdtv/linux_media #380: error when compiling

JOACO2303, 2025-02-26:
I followed the steps to install drivers correctly but when compiling this error appears:
make -C /usr/src/media_build/v4l 
make[1]: se entra en el directorio '/usr/src/media_build/v4l'
scripts/make_makefile.pl
./scripts/make_myconfig.pl
[ ! -f "./config-mycompat.h" ] && echo "/* empty config-mycompat.h */" > "./config-mycompat.h" || true
perl scripts/make_config_compat.pl /lib/modules/5.15.0-131-generic/build ./.myconfig ./config-compat.h
creating symbolic links...
make -C firmware prep
make[2]: Entering directory '/usr/src/media_build/v4l/firmware'
make[2]: Nothing to be done for 'prep'.
make[2]: Leaving directory '/usr/src/media_build/v4l/firmware'
make -C firmware
make[2]: Entering directory '/usr/src/media_build/v4l/firmware'
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/usr/src/media_build/v4l/firmware'
Kernel build directory is /lib/modules/5.15.0-131-generic/build
make -C ../linux apply_patches
make[2]: Entering directory '/usr/src/media_build/linux'
Syncing with dir ../../media
Patches for 5.15.0-131-generic already applied.
make[2]: Leaving directory '/usr/src/media_build/linux'
make -C /lib/modules/5.15.0-131-generic/build M=/usr/src/media_build/v4l  modules
make[2]: Entering directory '/usr/src/linux-headers-5.15.0-131-generic'
  CC [M]  /usr/src/media_build/v4l/mc-device.o
  CC [M]  /usr/src/media_build/v4l/mc-devnode.o
  CC [M]  /usr/src/media_build/v4l/mc-entity.o
  CC [M]  /usr/src/media_build/v4l/mc-request.o
  CC [M]  /usr/src/media_build/v4l/mc-dev-allocator.o
  LD [M]  /usr/src/media_build/v4l/mc.o
  CC [M]  /usr/src/media_build/v4l/v4l2-async.o
  CC [M]  /usr/src/media_build/v4l/v4l2-flash-led-class.o
  CC [M]  /usr/src/media_build/v4l/v4l2-fwnode.o
  CC [M]  /usr/src/media_build/v4l/v4l2-h264.o
  CC [M]  /usr/src/media_build/v4l/v4l2-jpeg.o
  CC [M]  /usr/src/media_build/v4l/v4l2-mem2mem.o
  CC [M]  /usr/src/media_build/v4l/v4l2-vp9.o
  CC [M]  /usr/src/media_build/v4l/tuner-core.o
  LD [M]  /usr/src/media_build/v4l/tuner.o
  CC [M]  /usr/src/media_build/v4l/v4l2-dv-timings.o
  CC [M]  /usr/src/media_build/v4l/v4l2-dev.o
  CC [M]  /usr/src/media_build/v4l/v4l2-ioctl.o
  CC [M]  /usr/src/media_build/v4l/v4l2-device.o
  CC [M]  /usr/src/media_build/v4l/v4l2-fh.o
  CC [M]  /usr/src/media_build/v4l/v4l2-event.o
  CC [M]  /usr/src/media_build/v4l/v4l2-subdev.o
  CC [M]  /usr/src/media_build/v4l/v4l2-common.o
  CC [M]  /usr/src/media_build/v4l/v4l2-ctrls-core.o
  CC [M]  /usr/src/media_build/v4l/v4l2-ctrls-api.o
  CC [M]  /usr/src/media_build/v4l/v4l2-ctrls-request.o
  CC [M]  /usr/src/media_build/v4l/v4l2-ctrls-defs.o
  CC [M]  /usr/src/media_build/v4l/v4l2-compat-ioctl32.o
  CC [M]  /usr/src/media_build/v4l/v4l2-mc.o
  CC [M]  /usr/src/media_build/v4l/v4l2-spi.o
  CC [M]  /usr/src/media_build/v4l/v4l2-trace.o
  CC [M]  /usr/src/media_build/v4l/v4l2-i2c.o
  LD [M]  /usr/src/media_build/v4l/videodev.o
  CC [M]  /usr/src/media_build/v4l/dvbdev.o
  CC [M]  /usr/src/media_build/v4l/dmxdev.o
  CC [M]  /usr/src/media_build/v4l/dvb_demux.o
  CC [M]  /usr/src/media_build/v4l/dvb_ca_en50221.o
  CC [M]  /usr/src/media_build/v4l/dvb_frontend.o
  CC [M]  /usr/src/media_build/v4l/dvb_net.o
  CC [M]  /usr/src/media_build/v4l/dvb_ringbuffer.o
  CC [M]  /usr/src/media_build/v4l/dvb_vb2.o
  CC [M]  /usr/src/media_build/v4l/dvb_math.o
  LD [M]  /usr/src/media_build/v4l/dvb-core.o
  CC [M]  /usr/src/media_build/v4l/rc-adstech-dvb-t-pci.o
  CC [M]  /usr/src/media_build/v4l/rc-alink-dtu-m.o
  CC [M]  /usr/src/media_build/v4l/rc-anysee.o
[... 392 lines cut ...]
      |                    ^~~~
/usr/src/media_build/v4l/tbs_pcie.c: In function 'tbs_pcie_audio_copy':
/usr/src/media_build/v4l/tbs_pcie.c:1667:8: error: implicit declaration of function 'copy_to_iter_fromio'; did you mean 'copy_to_user_fromio'? [-Werror=implicit-function-declaration]
 1667 |  ret = copy_to_iter_fromio(iter,runtime->dma_area+pos,bytes);
      |        ^~~~~~~~~~~~~~~~~~~
      |        copy_to_user_fromio
/usr/src/media_build/v4l/tbs_pcie.c: At top level:
/usr/src/media_build/v4l/tbs_pcie.c:1681:3: error: 'struct snd_pcm_ops' has no member named 'copy'
 1681 |  .copy =   tbs_pcie_audio_copy,
      |   ^~~~
/usr/src/media_build/v4l/tbs_pcie.c:1681:12: error: initialization of 'int (*)(struct snd_pcm_substream *, struct timespec64 *, struct timespec64 *, struct snd_pcm_audio_tstamp_config *, struct snd_pcm_audio_tstamp_report *)' from incompatible pointer type 'int (*)(struct snd_pcm_substream *, int,  long unsigned int,  struct iov_iter *, long unsigned int)' [-Werror=incompatible-pointer-types]
 1681 |  .copy =   tbs_pcie_audio_copy,
      |            ^~~~~~~~~~~~~~~~~~~
/usr/src/media_build/v4l/tbs_pcie.c:1681:12: note: (near initialization for 'tbs_capture_pcm_ops.get_time_info')
/usr/src/media_build/v4l/tbs_pcie.c: In function 'SignalDetectThread':
/usr/src/media_build/v4l/tbs_pcie.c:2001:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
 2001 |  int i;
      |  ^~~
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:297: /usr/src/media_build/v4l/tbs_pcie.o] Error 1
make[2]: *** [Makefile:1910: /usr/src/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.15.0-131-generic'
make[1]: *** [Makefile:53: default] Error 2
make[1]: se sale del directorio '/usr/src/media_build/v4l'
make: *** [Makefile:26: all] Error 2

marslosttemple, 2025-02-26:
I also meet this "copy_to_iter_fromio" error today, looks it is defined in pcm.h and linux/sound/core/memory.c

Davin622, 2025-02-27:
Hello, because the funcation is for new kernel version. so it will get the errors on old version. now we have withdrawn the update. please re-download the source code.

marslosttemple, 2025-02-27:
I re-downloaded the src code, yes, build OK now.

closed 2025-02-27 by crazycat69