tbsdtv/linux_media #383: Ubuntu 22.04 and 24.04 system instability

donCaster-tech, 2025-03-09:
dear all i'm going crazy about an big problemi with the tbs driver. I have some fixed pcs that works like a Tv recorders, I'm trying to install a newone with a routine that I always used, but after install of tbs drivers I have a lot of segfault errors in many processes like these:

```
[ 4208.871939] enc0:0:h264_nve[14658]: segfault at 18 ip 00007caed48a1743 sp 00007caebcff2d70 error 4 in libc.so.6[7caed4828000+195000] likely on CPU 1 (core 1, socket 0)
[ 7157.443184] dmx0:mov,mp4,m4[30335]: segfault at 18 ip 000073468b0a1743 sp 0000734653df44c0 error 4 in libc.so.6[73468b028000+195000] likely on CPU 1 (core 1, socket 0)
[ 7638.354651] dec0:1:h264_cuv[31655]: segfault at 18 ip 000078c115aa1743 sp 000078c0f33f3fe0 error 4 in libc.so.6[78c115a28000+195000] likely on CPU 1 (core 1, socket 0)
[ 7844.535306] dec0:1:h264_cuv[31667]: segfault at 18 ip 000076349dca1743 sp 00007634839f3fe0 error 4 in libc.so.6[76349dc28000+195000] likely on CPU 1 (core 1, socket 0)
[ 8348.411209] pool-tracker-mi[35614]: segfault at 18 ip 0000738c8fea1743 sp 0000738c899fe980 error 4 in libc.so.6[738c8fe28000+195000] likely on CPU 2 (core 2, socket 0)
[10514.803529] git-remote-http[36702]: segfault at 18 ip 000078a9758a1743 sp 00007ffd8c1f5170 error 4 in libc.so.6[78a975828000+195000] likely on CPU 0 (core 0, socket 0)
```

or these

```
[    7.556282] wireplumber[1305]: segfault at 10 ip 0000781c1c8ab7fc sp 00007fff9a07a0c0 error 4 in libc.so.6[ab7fc,781c1c828000+188000] likely on CPU 2 (core 2, socket 0)
[    9.763368] gnome-shell[1329]: segfault at 18 ip 00007fa34a0a9c21 sp 00007ffed9ca17a0 error 6 in libc.so.6[a9c21,7fa34a028000+188000] likely on CPU 0 (core 0, socket 0)
[    9.776181] colord-sane[1419]: segfault at 18 ip 00006ffd12aa9af7 sp 00007ffceab565d0 error 4 in libc.so.6[a9af7,6ffd12a28000+188000] likely on CPU 0 (core 0, socket 0)
[    9.824447] apport[1428]: segfault at 0 ip 0000000000590a1f sp 00007ffc55cebf70 error 4 in python3.12[190a1f,420000+2e3000] likely on CPU 2 (core 2, socket 0)
[  128.760054] apt[3003]: segfault at c ip 00007b4ed9cd1309 sp 00007ffdebbee7c0 error 4 in libapt-pkg.so.6.0.0[fc309,7b4ed9c0f000+161000] likely on CPU 2 (core 2, socket 0)
```

The problems are mainly triggered when hardware acceleration is in use for transcoding movies, but it is the whole system that becomes unstable.

If on the same machine, I install a new kernel version and switch to that, the errors disappear... until I reload the drivers. 

This happens on a machine with an intel core i5 6th generation processor that I am trying to configure from scratch. With kernel 6.8.0x and 6.11.x.

But the same problems also started on a machine that has always worked fine, on reboot after a few updates. 

Does anyone have any suggestions?

crazycat69, 2025-03-09:
what TBS hardware used ?

donCaster-tech, 2025-03-09:
TBS-6281 SE V2 on pc that is broken after update and quad tuner TBS-6205 SE on scratch install.

crazycat69, 2025-03-10:
> This happens on a machine with an intel core i5 6th generation processor that I am trying to configure from scratch. With kernel 6.8.0x and 6.11.x.
> But the same problems also started on a machine that has always worked fine, on reboot after a few updates.

Look like some kernel bug, i have similar issue with core i5-3470 (compiler crash when recompile drivers). First time i think this related to TBS 6314R driver (which destroy FPU/SIMD registers in kernel thread), but later this was fixed. And i not have this issue with 5.4, 5.15 kernels on ubuntu 20.04

donCaster-tech, 2025-03-11:
Probably yes, is a kernel problem...but I have the same problem also on a system with intel i9-9900K.
For the system with i5-6th I have installed Ubuntu 22.04 with ga kernel 5.15.0.134 and from yesterday works correctly. 
This morning same problem compare on Ubuntu 24.04 with kernel 6.11.0.17 and cpu intel i9-9900K, I try to install 6.11.0.19 and come back to 6.8.0.55 but the segfault problem during transcoding are always spresent. With 6.11.0.19 the system not load the TBS drivers.
Something changes into the kernel that conflict with TBS driver and nvidia-driver

donCaster-tech, 2025-03-13:
> I am using a TBS 5927 and I have the same problem on all 6.x kernel versions. It took me a while to find out, that the SegFaults only appear, when the TBS modules are loaded. And it does not depend on the pc hardware. No difference, if I test it on a consumer Notebook with AMD CPU or on a Fujitsu Server System with Intel CPU

Thanks for your feedback Jerome, not being the only one makes me happy :)

I confirm that the system is only stable with kernels below 6.x (in my case 5.15). Something must have changed in some component of the 6.x kernels because my systems have always worked fine. In my case the problem occurs when I try to install TBS drivers on a new kernel. Some machines are configured with hwe kernels (due to a misjudgment on my part) and cyclically update automatically on reboot. I always simply re-installed the driver, rebooted and off it went, the system was back up and running. This time I realised that if I reboot with the new kernel and install the drivers, the system starts to create a flood of segmentation errors. If I simply change the grub and reboot the system with the old kernel (without doing any operations), the system starts to work normally again. It is as if they have changed something in the compilers or some other component, so if you try to install the driver in the new kernel something goes wrong.

jp112sdl, 2025-03-13:
My problem is that my hardware is so new that I can't find a Linux distribution with kernel 5.15 that supports my network adapter. 

I tried to use the `-b master` branch instead of `-latest`.
But then I run into issue #209 
If my time allows, I will try to investigate further.

At the moment almost every python script crashes with "Segmentation fault".
Very annoying...

OldAVman, 2025-03-14:
You are certainly not alone.
As I prefer a simple OS for my media boxes I use a Fedora minimal install with X and openbox window manager.
I first encountered this problem when I upgraded an Intel 9th gen system which had been running fine for some time with kernel 6.12.9  to kernel 6.13.5. I simply upgraded the kernel rebooted and installed the TBS driver. After rebooting the tuner card (TBS6902) was found correctly and worked but I suddenly started getting random applications segfaulting and the journal flooded with errors exactly as you describe. As the 6.13 series kernels had a rather bad reputation I blamed this and reverted to 6.12.15 which also misbehaved in the same way. By now I had lost the original 6.12.9 from the GRUB menu so I  pulled a copy from Koji, installed it and added the driver. This now worked fine so I concluded that the problems only occurred with kernels above 6.12.9.
BUT
I later did a fresh OS install on a second Intel box (13100 processor) which came by default with the 6.13.5 kernel. I compiled and loaded the TBS driver fully expecting problems but got none. It has now worked perfectly for several days with no segfaults or other errors.
Clearly the driver CAN work with the latest kernels but as I use the same procedure for updating kernels, compiling and loading the driver I can't explain why success seems to be totally random.  Any explanations for a possible cause would be most welcome!

tasoss, 2025-03-16:
I had this issue on various ubuntu machines with different tbs cards. All of them had this problem (more than 15 computers).
I REALLY don't know how they work right now( i compiled again the drivers , installed updates-which were up to date). 
Yesterday , after an update , in one of them the problem appeared again.
NO 6.x series is NOT the problem. I am using the drivers on such kernels( ubuntu and Slackware) with no problems.
So, i get segmentation faults even when i was trying to compile the kernel. After trying 2-3 times ,  finally i managed to compile the kernel. I think the problem has to do with some ubuntu specific update on some library but i really don't know what is going on and it is really disturbing. By luck, except this one machine , all the other is working as expected.
I will certainly notify in case i find something but it would be nice to let us know, in case someone finds what is going on.
Right now i have this one machine to experiment but i don't know for how much because it's a production machine.

OldAVman, 2025-03-16:
I don't think this is Ubuntu specific, as I reported above I'm using a very basic OS derived from a Fedora minimal install. My systems are Intel based using the internal intel graphics. The only non standard addition to the kernel is the TBS driver. In my failed installations I didn't have a problem compiling and installing the driver which seemed to go smoothly. The tuners were found and worked correctly but then during usage I started to get random crashes in various seemingly unrelated programs. On examining dmesg and the journal there were many error messages similar to those reported by donCaster-tech.

jp112sdl, 2025-03-16:
I agree qith @OldAVman.
It's not Ubuntu specific. 
I also had the SegFault problems in Debian and Linux Mint

tasoss, 2025-03-16:
I have/had the same issue. A program is using the adapters. This same program calls external applications(ffmpeg, mediainfo etc) . Those programs crash. The fact that i had segfaults during compilation of apps/kernel is something i observed in the latest computer i have seen the problem appear, as i had mentioned above.
I don't use internal Intel graphics but all the cpus are Intel.
Of course it may not be Ubuntu specific. It is just that all the computers(in my case) are ubuntu based. Except the Slackware ones which don't have any problems (using 6.x kernels).
Moreover i tried previous versions of the TBS drivers and the problem was there.

LadyChloe, 2025-03-16:
I have updated Arch Linux, which includes the kernel version 6.13.7.arch1-1. All seemed well, until I installed the latest TBS drivers from git (I have TBS6903-X). Random software started creating core dumps, including systemd, udevadm, find, bash, zsh and most often python which processes satellite meteorological data from Eumetcast.
I have tried installing the LTS kernel, version 6.12.19-1 but it also had segfaults with the latest TBS drivers. I tried building the TBS drivers from October on this kernel (which I knew worked back then) but I did not manage to due to some patch problems and I did not explore this further.
In the end I restored the old kernel 6.10.3-arch1-2 along with all modules, which includes TBS drivers from that time (Oct 20, 2024). The system works flawlessly with this. It's a low power Intel N100 system.
This suggests that 6.x kernels are not the problem but rather the newer TBS driver version. I had been running TBS on 6.x for quite some time, until this problem happened.

tasoss, 2025-03-16:
I had used specific git commit ids (that i knew the driver was working) and again the problem was there. To be honest it took longer to sigfault and i was sure i had solved the problem. But then it appeared again.
So i wish i could say with no doubt that it's a driver's problem. But i can't.
Of course , using an older kernel is not a proper solution ,at least, for me.
I will try to investigate more.

LadyChloe, 2025-03-16:
The one that's working for me is at commit 06f0fc5ccc29edd1ba0378087fc044c0b63a46d1
I have been using it since 20 Oct without a single segfault.

fe31nz, 2025-03-18:
I am using a Ubuntu 24.04 system on 6.8 kernels with TBS6909 and TBS6209 tuner cards for MythTV.  For the past few weeks it has been doing SYSSEGV and SYSABRT crashes in lots of different programs, including mythbackend and minisatip.  For all the crash dumps I have looked at, the crashes appear to be in the system library malloc routines, either with an access outside the allocated RAM (SYSSEGV) or the malloc routine seeing bad data (a corrupted pointer for example) and killing the program with SIGABRT.  Last night I wanted to see if I could build an older version of the TBS drivers that did not have this bug, but I was unable to as various commands called by the build process died on SIGSEGV or SIGABRT and stopped or corrupted the build.  Previously I have only done a build of the TBS drivers with the standard kernel drivers in place, after a reboot when new kernel was installed.  So it looks like I will need to delete the TBS drivers and restore the standard kernel ones before I can do a new TBS drivers build.  Something like this:

 rm -rf /lib/modules/`uname -r`/kernel/drivers/media/*
apt install --reinstall linux-modules-extra-`uname -r`
reboot

Then I need to know how to build an old version of the TBS drivers that will work with the 6.8 kernels.  I have never learnt much about git, so that is a bit of a problem.  Is there anyone who can provide me details on how to do it?

This is my current build script:

`#!/bin/bash

set -x
cd /root/projects/tbs6909-drivers || exit 1
rm -r media
rm -r media_build
git clone https://github.com/tbsdtv/media_build.git || exit 2
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media || exit 3

cd media || exit 9
patch -p0 <../mxl58x.patch
if [ $? -ne 0 ]; then
    cd ..
    exit 10
fi
cd ..

cd media_build

## Go back to a version that does not do memory corruption (SIGSEGV, SIGABRT)???
#git reset --hard 0f49c76b80838ded04bd64c56af9e1f9b8ac1965

sed -i "s/add v6.8-ccs.patch/#add v6.8-ccs.patch/" backports/backports.txt || exit 11
make cleanall || exit 12
make dir DIR=../media || exit 13
./install.sh || exit 14`

The commented out "git reset" line is my first attempt at getting an older version of the TBS driver code.  Is that the sort of thing I should be doing?

crazycat69, 2025-03-18:
Have some idea, need testing
https://github.com/tbsdtv/media_build/commit/f508ea49e8678265e6926d6e1b2f349c2e07270f

tasoss, 2025-03-18:
@crazycat69 Well that was quick. It doesn't help.
I was using drivers ,from 2024 commits ,  for almost one day , nothing happened.
I have just rebuilt from latest (with the new commit- both media and build) and it sigfaults.

tasoss, 2025-03-19:
It seems to be working correctly but i would like to test it a few more hours.
Great job!

crazycat69, 2025-03-20:
Look like all media modules was compiled with FPU/SIMD support and this corrupt user-space FPU/SIMD registers which cause segfaults. Now some tbscapture2 sources compiled with FPU/SIMD and registers protected (TBS 6314R PCIEx4 capture card support).

tasoss, 2025-03-20:
I was going to ask you after the completion of my tests.
How did you manage to find that this was the bug ?
BTW why in Slackware (Linux distribution) i didn't have this problem ? Any ideas ?
Thanks again!

crazycat69, 2025-03-20:
Maybe user-space tools not use FPU/SIMD.
For kernel modules FPU/SIMD default disabled (some crypto modules use it, but with register protection).

tasoss, 2025-03-20:
> Cat sense, no more :)

Lollll

tasoss, 2025-03-20:
As far as i am concerned , i would say it is fixed.

OldAVman, 2025-03-20:
Tested on the box where I'd previously had to downgrade the kernel in order to successfully install the driver. Updated the kernel to the newest Fedora version (6.13.7), compiled and installed the new driver and everything is working fine. It has been running most of the day with no errors in the journal.
Looks good to me.
Thanks yet again Crazycat.

donCaster-tech, 2025-03-21:
at the moment I have all the machines in production, but soon I will have to configure one from scratch and test the solution. However, I am sure it will work. I thank @crazycat69 and all of you for actively contributing to solving problem that would have been very serious for me, you all have a paid beer :)

jp112sdl, 2025-03-22:
Today I was finally able to update my Debian system to kernel 6.12.12 and compile the TBS drivers with the latest commits.

No more SegFaults... Cheers @crazycat69 👍

closed 2025-03-22 by crazycat69