tbsdtv/linux_media #387: Raspberry PI 5 TBS 6902 - compile fails

andycockers, 2025-03-16:
CC [M]  /home/andy/blah/media_build/v4l/solo6x10-v4l2-enc.o
gcc-12: fatal error: Killed signal terminated program cc1
compilation terminated.
make[4]: *** [/usr/src/linux-headers-6.6.74+rpt-common-rpi/scripts/Makefile.build:248: /home/andy/blah/media_build/v4l/solo6x10-p2m.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/linux-headers-6.6.74+rpt-common-rpi/Makefile:1946: /home/andy/blah/media_build/v4l] Error 2
make[2]: *** [/usr/src/linux-headers-6.6.74+rpt-common-rpi/Makefile:246: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.6.74+rpt-rpi-2712'
make[1]: *** [Makefile:52: default] Error 2
make[1]: Leaving directory '/home/andy/blah/media_build/v4l'
make: *** [Makefile:26: all] Error 2
V4L drivers installing...
make -C /home/andy/blah/media_build/v4l install
make[1]: Entering directory '/home/andy/blah/media_build/v4l'
Installing kernel modules under /lib/modules/6.6.74+rpt-rpi-2712/updates/extra/media/:
/sbin/depmod -a 6.6.74+rpt-rpi-2712 
make -C firmware install
make[2]: Entering directory '/home/andy/blah/media_build/v4l/firmware'
Installing firmwares at /lib/firmware: 
make[2]: Leaving directory '/home/andy/blah/media_build/v4l/firmware'
make[1]: Leaving directory '/home/andy/blah/media_build/v4l'
V4L drivers installation done
You need to reboot...

The compile failed, so there's no /dev/dvb device after a reboot.

I'm using the offline driver package from https://www.tbsiptv.com/download/common/tbs-open-linux-drivers_v20240829.zip

crazycat69, 2025-03-16:
how many memory installed ? swap enabled ?
maybe need run make on less cpu cores

andycockers, 2025-03-17:
8gb memory.
swap is enabled.

crazycat69, 2025-03-18:
https://github.com/tbsdtv/media_build/commit/f508ea49e8678265e6926d6e1b2f349c2e07270f