tbsdtv/linux_media #388: Raspberry PI 5 - tbsdtv forked driver - compile fails

andycockers, 2025-03-16:
```
CC [M]  /home/andy/github/tbsdriver/media_build/v4l/igorplugusb.o
/home/andy/github/tbsdriver/media_build/v4l/gpio-ir-recv.c:205:19: error: initialization of 'int (*)(struct platform_device *)' from incompatible pointer type 'void (*)(struct platform_device *)' [-Werror=incompatible-pointer-types]
  205 |         .remove = gpio_ir_recv_remove,
      |                   ^~~~~~~~~~~~~~~~~~~
/home/andy/github/tbsdriver/media_build/v4l/gpio-ir-recv.c:205:19: note: (near initialization for 'gpio_ir_recv_driver.remove')
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-6.6.74+rpt-common-rpi/scripts/Makefile.build:248: /home/andy/github/tbsdriver/media_build/v4l/gpio-ir-recv.o] Error 1
make[4]: *** Waiting for unfinished jobs....
/home/andy/github/tbsdriver/media_build/v4l/ir-hix5hd2.c:397:19: error: initialization of 'int (*)(struct platform_device *)' from incompatible pointer type 'void (*)(struct platform_device *)' [-Werror=incompatible-pointer-types]
  397 |         .remove = hix5hd2_ir_remove,
      |                   ^~~~~~~~~~~~~~~~~
/home/andy/github/tbsdriver/media_build/v4l/ir-hix5hd2.c:397:19: note: (near initialization for 'hix5hd2_ir_driver.remove')
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-6.6.74+rpt-common-rpi/scripts/Makefile.build:248: /home/andy/github/tbsdriver/media_build/v4l/ir-hix5hd2.o] Error 1
make[3]: *** [/usr/src/linux-headers-6.6.74+rpt-common-rpi/Makefile:1946: /home/andy/github/tbsdriver/media_build/v4l] Error 2
make[2]: *** [/usr/src/linux-headers-6.6.74+rpt-common-rpi/Makefile:246: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.6.74+rpt-rpi-2712'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/andy/github/tbsdriver/media_build/v4l'
make: *** [Makefile:26: all] Error 2
```

Raspberry PI 5 with Raspberry PI OS Bookworm

crazycat69, 2025-03-16:
[tbsdtv/media_build](https://github.com/tbsdtv/media_build) updated, please check

https://github.com/tbsdtv/media_build/commit/217946511997b27799a14dda4f46e76d022e1047

andycockers, 2025-03-17:
@crazycat69 thank you for the prompt reply.

I got the latest, as you suggested. It gets a bit further but then fails with:

```
In file included from /home/andy/github/tbsdriver/media_build/v4l/tbs_pcie2.c:8:
/home/andy/github/tbsdriver/media_build/v4l/tbs_pcie2.h:12:10: fatal error: asm/fpu/api.h: No such file or directory
   12 | #include <asm/fpu/api.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[4]: *** [/usr/src/linux-headers-6.6.74+rpt-common-rpi/scripts/Makefile.build:248: /home/andy/github/tbsdriver/media_build/v4l/tbs_pcie2.o] Error 1
make[4]: *** Waiting for unfinished jobs....
/home/andy/github/tbsdriver/media_build/v4l/compare.c: In function 'SumSquareErrorToPsnr':
/home/andy/github/tbsdriver/media_build/v4l/compare.c:274:8: error: '-mgeneral-regs-only' is incompatible with the use of floating-point types
  274 | double SumSquareErrorToPsnr(uint64_t sse, uint64_t count) {
      |        ^~~~~~~~~~~~~~~~~~~~
/home/andy/github/tbsdriver/media_build/v4l/compare.c: In function 'CalcFramePsnr':
/home/andy/github/tbsdriver/media_build/v4l/compare.c:291:8: error: '-mgeneral-regs-only' is incompatible with the use of floating-point types
  291 | double CalcFramePsnr(const uint8_t* src_a,
      |        ^~~~~~~~~~~~~
/home/andy/github/tbsdriver/media_build/v4l/compare.c: In function 'I420Psnr':
/home/andy/github/tbsdriver/media_build/v4l/compare.c:304:8: error: '-mgeneral-regs-only' is incompatible with the use of floating-point types
  304 | double I420Psnr(const uint8_t* src_y_a,
      |        ^~~~~~~~
/home/andy/github/tbsdriver/media_build/v4l/compare.c: In function 'CalcFrameSsim':
/home/andy/github/tbsdriver/media_build/v4l/compare.c:389:8: error: '-mgeneral-regs-only' is incompatible with the use of floating-point types
  389 | double CalcFrameSsim(const uint8_t* src_a,
      |        ^~~~~~~~~~~~~
/home/andy/github/tbsdriver/media_build/v4l/compare.c: In function 'Ssim8x8_C':
/home/andy/github/tbsdriver/media_build/v4l/compare.c:335:15: error: '-mgeneral-regs-only' is incompatible with the use of floating-point types
  335 | static double Ssim8x8_C(const uint8_t* src_a,
      |               ^~~~~~~~~
/home/andy/github/tbsdriver/media_build/v4l/compare.c: In function 'I420Ssim':
/home/andy/github/tbsdriver/media_build/v4l/compare.c:418:8: error: '-mgeneral-regs-only' is incompatible with the use of floating-point types
  418 | double I420Ssim(const uint8_t* src_y_a,
      |        ^~~~~~~~
make[4]: *** [/usr/src/linux-headers-6.6.74+rpt-common-rpi/scripts/Makefile.build:248: /home/andy/github/tbsdriver/media_build/v4l/compare.o] Error 1
make[3]: *** [/usr/src/linux-headers-6.6.74+rpt-common-rpi/Makefile:1946: /home/andy/github/tbsdriver/media_build/v4l] Error 2
make[2]: *** [/usr/src/linux-headers-6.6.74+rpt-common-rpi/Makefile:246: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.6.74+rpt-rpi-2712'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/andy/github/tbsdriver/media_build/v4l'
```

crazycat69, 2025-03-17:
https://github.com/tbsdtv/linux_media/commit/a8b05e2cf2520df776abdf6c918458852a5e5398

andycockers, 2025-03-17:
Thank you. It gets further, then fails with:

```
  CC [M]  /home/andy/github/tbsdriver/media_build/v4l/solo6x10-v4l2-enc.o
gcc-12: fatal error: Killed signal terminated program cc1
compilation terminated.
make[4]: *** [/usr/src/linux-headers-6.6.74+rpt-common-rpi/scripts/Makefile.build:248: /home/andy/github/tbsdriver/media_build/v4l/solo6x10-v4l2-enc.o] Error 1
make[4]: *** Waiting for unfinished jobs....
```

andycockers, 2025-03-18:
Limiting the cores to 2 allowed it to complete

closed 2026-01-21 by Davin622