tbsdtv/linux_media #390: Failed compilation on kernel 6.14

alfrup, 2025-04-06:
I need to compile a driver for a TBS6590 device. When I execute make -j4 I get the following errors:

tbs_pcie2.c: In function 'tbs_pcie_audio_trigger':
tbs_pcie2.c:1637:27: warning: unused variable 'chip' [-Wunused-variable]
 1637 |         struct tbs_audio *chip = snd_pcm_substream_chip(substream);
      |                           ^~~~
tbs_pcie2.c: In function 'tbs_pcie_audio_copy':
tbs_pcie2.c:1680:57: error: passing argument 2 of 'copy_to_iter_fromio' makes integer from pointer without a cast [-Wint-conversion]
 1680 |         ret = copy_to_iter_fromio(iter,runtime->dma_area+pos,bytes);
      |                                        ~~~~~~~~~~~~~~~~~^~~~
      |                                                         |
      |                                                         unsigned char *
In file included from tbs_pcie2.h:25,
                 from tbs_pcie2.c:8:
/usr/src/linux-6.14.0-1/include/sound/pcm.h:1535:60: note: expected 'size_t' {aka 'long unsigned int'} but argument is of type 'unsigned char *'
 1535 | size_t copy_to_iter_fromio(const void __iomem *src, size_t bytes,
      |                                                     ~~~~~~~^~~~~
tbs_pcie2.c:1680:62: error: passing argument 3 of 'copy_to_iter_fromio' makes pointer from integer without a cast [-Wint-conversion]
 1680 |         ret = copy_to_iter_fromio(iter,runtime->dma_area+pos,bytes);
      |                                                              ^~~~~
      |                                                              |
      |                                                              long unsigned int
/usr/src/linux-6.14.0-1/include/sound/pcm.h:1536:45: note: expected 'struct iov_iter *' but argument is of type 'long unsigned int'
 1536 |                            struct iov_iter *iter) __must_check;
      |                            ~~~~~~~~~~~~~~~~~^~~~
make[5]: *** [/usr/src/linux-6.14.0-1/scripts/Makefile.build:208: tbs_pcie2.o] Fehler 1
make[5]: *** Es wird auf noch nicht beendete Prozesse gewartet …
make[4]: *** [/usr/src/linux-6.14.0-1/Makefile:2017: .] Fehler 2
make[3]: *** [/usr/src/linux-6.14.0-1/Makefile:251: __sub-make] Fehler 2
make[3]: Verzeichnis „/usr/local/sbin/media_build/v4l“ wird verlassen
make[2]: *** [../../../linux-6.14.0-1/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-6.14.0-1-obj/x86_64/default'
make[1]: *** [Makefile:53: default] Fehler 2
make[1]: Verzeichnis „/usr/local/sbin/media_build/v4l“ wird verlassen
make: *** [Makefile:26: all] Fehler 2

crazycat69, 2025-04-06:
https://github.com/tbsdtv/linux_media/commit/b4a40fd7cc38e679da00859f5d235e88af15b41a

alfrup, 2025-04-07:
> Antwort an: tbsdtv/linux_media
> (Issue #390)
>
>  b4a40fd [1]
> [3].
> crazycat69crazycat69
> b4a40fd [1]
> [3].
>
> [1] b4a40fd
>     https://github.com/tbsdtv/linux_media/commit/b4a40fd7cc38e679da00859f5d235e88af15b41a
> [2] view it on GitHub
>     https://github.com/tbsdtv/linux_media/issues/390#issuecomment-2781487259
> [3] unsubscribe

Thank you, it works.