tbsdtv/linux_media #391: TVheadend and TBS5930 - Video Stuttering

PrvtN00b, 2025-04-10:
After I compile and run everything from scratch tvheadend and the tbs5930 works perfect without any single problem, but after a few hours of using it the problem starts, in my micro tiny server (hp chrombox G2 with 16gb of ram) with Ubunbuntu 20(also tested with debian 12) I only have CasaOS running, tvheadend installed as a docker compose.

[image]

[image]

compilation steps followed:

> root@debian:~# uname -r
> 6.12.9+bpo-amd64
> 
> lsusb -vvv | grep 5930
> 
> apt install linux-headers-$(uname -r)
> apt install libproc-processtable-perl patchutils gcc build-essential git libdigest-sha-perl build-essential subversion mercurial openssl gettext libssl-dev screen libv4l-dev flex bison -y 
> git clone https://github.com/tbsdtv/media_build.git
> git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
> cd media_build
> make dir DIR=../media
> make distclean
> make allyesconfig
> sed -i -r 's/(^CONFIG.*_RC.*=)./\1n/g' v4l/.config
> sed -i -r 's/(^CONFIG.*_IR.*=)./\1n/g' v4l/.config
> make -j2
> sudo make install
> wget http://www.tbsdtv.com/download/document/linux/tbs-tuner-firmwares_v1.0.tar.bz2
> sudo tar jxvf tbs-tuner-firmwares_v1.0.tar.bz2 -C /lib/firmware/

crazycat69, 2025-04-10:
maybe overheating ?

smp79, 2025-08-10:
I think this could be related - https://github.com/tbsdtv/linux_media/issues/410#issuecomment-3172901860