tbsdtv/linux_media #392: Add support for TV tuner DVB-T2 AVerTV Mobile 510 (EW510)

Vyacheslav-S, 2025-04-15:
**Add support for TV tuner DVB-T2 AVerTV Mobile 510 (EW510).**

AVerTV Mobile 510 is a compact TV tuner, the dimensions of the device, excluding the microUSB connector, are 42×22×10 mm. with a weight of only 7.5 g.
The Si2147 RF module and the Silicon Laboratories DVB-T2-enabled Si2168 demodulator are responsible for signal reception and processing, while the IT9303 USB bridge provides interface communication.

[image]
[image]
[image]

**Patch:**

```
--- /home/svm/Work/AVerTV Mobile 510/linux-media/af9035.c
+++ /home/svm/Work/AVerTV Mobile 510/linux-media/af9035_new.c
@@ -860,7 +860,9 @@
 		 * configurations seen so far.
 		 */
 		if ((le16_to_cpu(d->udev->descriptor.idVendor) == USB_VID_AVERMEDIA) &&
-		    (le16_to_cpu(d->udev->descriptor.idProduct) == USB_PID_AVERMEDIA_TD310)) {
+		    (le16_to_cpu(d->udev->descriptor.idProduct) == USB_PID_AVERMEDIA_TD310) ||
+		    (le16_to_cpu(d->udev->descriptor.idVendor) == USB_VID_AVERMEDIA) &&
+			(le16_to_cpu(d->udev->descriptor.idProduct) == 0x0510)) {
 			state->it930x_addresses = 1;
 		}
 		return 0;
@@ -1637,6 +1639,8 @@
 	 */
 	if ((le16_to_cpu(d->udev->descriptor.idVendor) == USB_VID_DEXATEK &&
 	     le16_to_cpu(d->udev->descriptor.idProduct) == 0x0100) ||
+	     (le16_to_cpu(d->udev->descriptor.idVendor) == USB_VID_AVERMEDIA &&
+	     le16_to_cpu(d->udev->descriptor.idProduct) == 0x0510) ||
 	    (le16_to_cpu(d->udev->descriptor.idVendor) == USB_VID_TERRATEC &&
 	     le16_to_cpu(d->udev->descriptor.idProduct) == USB_PID_TERRATEC_CINERGY_TC2_STICK))
 		si2157_config.dont_load_firmware = true;
@@ -2156,6 +2160,8 @@
 		&it930x_props, "Logilink VG0022A", NULL) },
 	{ DVB_USB_DEVICE(USB_VID_TERRATEC, USB_PID_TERRATEC_CINERGY_TC2_STICK,
 		&it930x_props, "TerraTec Cinergy TC2 Stick", NULL) },
+	{ DVB_USB_DEVICE(USB_VID_AVERMEDIA, 0x0510,
+		&it930x_props, "AVerMedia Mobile-T2", NULL) },
 	{ }
 };
 MODULE_DEVICE_TABLE(usb, af9035_id_table);
```

crazycat69, 2025-04-15:
Merged

Vyacheslav-S, 2025-04-15:
Great!

closed 2025-04-15 by Vyacheslav-S