tbsdtv/linux_media #394: TBS6209SE - Transport error indicator

xeonmasters, 2025-04-18:
Hello,

I am using the last crazycat69 driver (2025-04-18), i have made a fresh install on a ubuntu 22.04 server with the tune script (https://cdn.cesbo.com/astra/scripts/tune.sh) and I got no good results

I have tried to compile the driver with enable_msi = false (on linux_media/drivers/media/pci/tbsecp3/tbsecp3-core.c), and I have tried to change tbs.conf/tbsecp3.conf to:

```
"options tbsecp3 int_type=1
options tbsecp3 enable_msi=0"
```

And on dmesg:

```
[    5.060007] TBSECP3 driver 0000:01:00.0: enabling device (0000 -> 0002)
[    5.060100] TBSECP3 driver 0000:01:00.0: Turbosight TBS 6209SE(Octa DVB-T/T2/C/C2/ISDB-T/C/ATSC)
[    5.068047] TBSECP3 driver 0000:01:00.0: MSI disabled by module parameter 'enable_msi'
```

But no good results were seen. I am currently using it with a DVB-C provider. All I see on tvheadend logs is:

[image]

But signal is good and BER is 0

[image]

In the same dedicated, I have tested with TBS 6209 (Octa DVB-C) and was fine (same frequency, same channel, exacly same tvheadend and same driver). So, thats no issue with my hardware - and no errors have been seen on dmesg as well

[image]

Linux version is: Ubuntu 22.04.5 LTS - Kernel: Linux tvh-4 5.15.0-138-generic

Thanks

xeonmasters, 2025-04-18:
I believe the problem is related to the TBS amplifier, is there any way to disable it?

crazycat69, 2025-04-18:
Enable extended signal statistic in tvh (RFLevel in dBm, SNR in dB), but i think you right about LNA in 6902SE. So you need some RF attenuator or tap before 6902SE

xeonmasters, 2025-04-22:
Dear @crazycat69 thanks for your answer! 

- extended signal statistic in tvh: are those available in tvheadend 4.3.0? I havent found this option

- RF attenuator or tap - After testing with a High Pass Filter (HPF54) it solved, no transport errors at start - but on some channels, it gets some CC error's, I am running the same channel at same time on 2 TBS (TBS6209SE and TBS6209), but only 6209SE get those errors

Is there a way to disable the amplifier on TBS6209se? I will test with a TAP soon and report

Again, thank you very much

xeonmasters, 2025-04-24:
"Extended signal statistic" I haven't found this option, but I have tested in two tuners with "Force old status" enabled and disabled.

[image]

And this is how it was before using the -10dB TAP:

[image]

Even with the signal attenuator (and HPF-54 filter), continuity errors appear sporadically (like 1~5 CC errors every hour), unlike the TBS6209 (testing at same time). but certainly, the problem is related to LNA.

Thanks for the help!