tbsdtv/linux_media #398: m88rs6060 - wrong ts speed when tuning higher order dvb-s2x modcodes

majortom9, 2025-05-17:
Noticed when tuning a dvb-s2x TP having SR 34285 16apsk FEC 25/36 that the Transport Stream
was only about 1/2 what it should be (only 50 some odd Mbps). After troubleshooting the issue led me to this fix in 
m88rs6060 module.  In m88rs6060_set_clock_ratio() it was falling through the if block to DVB-S, which
only has mod_fac of 2 (qpsk), thus why the speed was only 1/2 what it should have been.
Below clocks the TS in at the right rate now:

`diff --git a/drivers/media/dvb-frontends/m88rs6060.c b/drivers/media/dvb-frontends/m88rs6060.c
index a50cdefcb375..00034a060b76 100644
--- a/drivers/media/dvb-frontends/m88rs6060.c
+++ b/drivers/media/dvb-frontends/m88rs6060.c
@@ -2429,7 +2429,7 @@ static int m88rs6060_set_clock_ratio(struct m88rs6060_dev *dev )
 
        m88rs6060_get_channel_info(dev,&p_info);
 
-       if(p_info.type==MtFeType_DvbS2){
+       if(p_info.type==MtFeType_DvbS2 || p_info.type == MtFeType_DvbS2X){
                dev_dbg(&i2c->dev, "iPlsCode = 0x%02X\n", p_info.iPlsCode);
                switch(p_info.mod_mode)
                {
`

crazycat69, 2025-05-17:
Fixed, thanks.
https://github.com/tbsdtv/linux_media/commit/bc9c6ba3b5c1638d8e505fd650091adccd14429c

majortom9, 2025-05-17:
awesome, thx crazycat... 
sidenote, I was originally looking in to that after a report from a windows user with tbs5930 that they couldn't play that mux. Can't say is similar issue because I don't use windows at all. Just that there may be an issue with it in Windows.
Might be worth a look, if ur doing any windows development with 5930.

crazycat69, 2025-05-18:
Yes, all drivers use common code from Montage Tech with this bug. But this TBS PCI-E devices with this chip use auto TS clock, so this function not used.

majortom9, 2025-05-18:
hmm, tbs5930 is a USB interface. Does that support the auto TS clock ur speaking of in windows?

crazycat69, 2025-05-18:
need check it :) add **m88rs6060_config.ts_autoclock=1** to [tbs5930_frontend_attach()](https://github.com/tbsdtv/linux_media/blob/latest/drivers/media/usb/dvb-usb/tbs5930.c#L148C12-L148C35)

crazycat69, 2025-05-18:
i just review original Montage tech code and windows driver sources - **mt_fe_dmd_rs6060_set_clock_ratio()** have some another code:

```
	mt_fe_dmd_rs6060_get_channel_info(handle, &p_info);

	if (handle->tp_cfg.mCurrentType == MtFeType_DvbS2)			// for dvb-s2
	{
		switch(p_info.mod_mode)

```

**mt_fe_dmd_rs6060_get_channel_info()** set **handle->tp_cfg.mCurrentType** _MtFeType_DvbS_ or _MtFeType_DvbS2_ - so this only linux driver issue.

majortom9, 2025-05-18:
But what happens when MtFeType_Dvbs2x and not MtFeTpe_Dvbs2?
On this s2x TP, it will similarly fall thru that if block, won't it?

majortom9, 2025-05-18:
I had the windows user show me the data rate from their tbs5930 while tuned to that troublesome mux and it looks correct to me.
Same as what I get in linux, after the fix action. So their issue must be something else. Waiting on a .ts file to verify.
Not being familiar with the windows driver, I asked what version they had installed, and it's 'v1007_20240605'
Requested the user try the Win11 driver, 'tbs5930_release_v11007_20250509_win11.zip' see what happens.

crazycat69, 2025-05-18:
Look like same driver compiled with MS VS2022, not work with Win7/8 anymore.

majortom9, 2025-05-24:
Thanks Crazycat. If ya get a chance to have a look thru the windows code again.
Some more detail...

A different windows user, who just bought a new TBS5930 this week, recorded a Full TS File for me from Alt DVB today.
Had them tune 99w 4120 V 32485 FEC 25/36 this time. Same provider as the mux I was working with. And sent me the file.

TSReader Playing back the file says the stream full stream is ~ 70 Mbps. And each program is only 7 Mbps. The
files I capture in linux, which we fixed, shows 94.1 Mbps for the full stream, and ~ 10 Mbps for each program.
roughly what the DVB-S2x Calculator says it should be.

[url=https://ibb.co/wfmVxQ2][img]https://i.ibb.co/1kW1wX4/abc-99w-4120-v-34285-ts.png[/img][/url]

So, checking the DVB-S2x calculator, for theoretical bitrates, and looking for possible combinations that would result in such a low
bitrate (in the case of windows driver bug(s)). There are two possibilities I see right off the bat. Either one of these two
scenarios could result in an overall bitrate of 70 Mbps.

1.) Locks it fine, but some other function() thinks it's an 8psk signal, so it gets clocked in wrong, similar to what the linux module was doing originally.
[img]https://i.ibb.co/67L8cmvB/if-8psk-fec-25-36-instead-of-16apsk-fec-25-36.png[/img]

2.) Locks it fine, but some other function() thinks it's a 16apsk signal, but having the 8_15 FEC instead of the 25_36 FEC. Remembering the other linux
module bug we found couple years ago, had exactly that problem. Locked 16apsk fec 25_36 fine, but reported it as fec 8_15.
My initial suspicion is possibility #2.

[img]https://i.ibb.co/Hf19wPgp/if-16apsk-fec-8-15-instead-of-fec25-36.png[/img]

crazycat69, 2025-05-24:
So maybe this VCM transponder with 16APSK, 25/36 and 8PSK/8/15 frames ?

crazycat69, 2025-05-24:
https://github.com/tbsdtv/linux_media/commit/d94513e97be429847a8bf6ba6bcf74492d3d24d9

Look like this not fixed in windows drivers, but fixed in my StreamReaderEx-BDA

majortom9, 2025-05-25:
Good News, we have a report of Windows now playing that mux perfectly with ur debug driver, like linux.
I think we can consider these bugs sqaushed now, like a fly in a venus flytrap.
Thanks again Crazycat,

majortom9, 2025-05-25:
For the record, windows user snagged an new TS file from the mux with the windows debug driver, and it looks good to me now.
They did mention they took a few hits in signal while recording, but it's a 16 apsk signal, so not surprising to me if marginal signal
quality. It's play perfectly for em now though. 
https://i.ibb.co/M5JZk0b2/Screenshot-at-2025-05-24-22-10-41.png

majortom9, 2025-05-25:
Here is another small bug I found today.
When tuning a mux of DVB-S QPSK FEC 7/8, is indicating FEC NONE.
Because it's falling through to default in m88rs6060_get_frontend().
In that function FEC 7_8 isn't listed.
This fixed it for me in linux, on my rpi. I haven't committed it to my local tree yet, but is definitely working now.
This was just an indication type issue, and wasn't affecting the performance, as ya could still
play the programs fine. Just we should definitely be able to indicate the correct FEC.
A windows user tells me they have the same issue when tuning the same DVB-S QPSK FEC 7/8  
muxes I was tuning. So may wanna review that on the windows side as well.

```
diff --git a/drivers/media/dvb-frontends/m88rs6060.c b/drivers/media/dvb-frontends/m88rs6060.c
index 49b457982e64..30cf5aaadf86 100644
--- a/drivers/media/dvb-frontends/m88rs6060.c
+++ b/drivers/media/dvb-frontends/m88rs6060.c
@@ -2449,6 +2449,7 @@ static int m88rs6060_get_frontend(struct dvb_frontend *fe, struct dtv_frontend_p
                        case MtFeCodeRate_3_4:          p->fec_inner = FEC_3_4; break;
                        case MtFeCodeRate_4_5:          p->fec_inner = FEC_4_5; break;
                        case MtFeCodeRate_5_6:          p->fec_inner = FEC_5_6; break;
+                       case MtFeCodeRate_7_8:          p->fec_inner = FEC_7_8; break;
                        case MtFeCodeRate_8_9:          p->fec_inner = FEC_8_9; break;
                        case MtFeCodeRate_9_10: p->fec_inner = FEC_9_10;        break;
                        case MtFeCodeRate_5_9:          p->fec_inner = FEC_5_9; break;
```

crazycat69, 2025-05-25:
https://github.com/tbsdtv/linux_media/commit/b00b2678fc0a23cf4956f33a84d50bde1d549fea

majortom9, 2025-05-26:
What's your opinion about tossin a crude m88rs6060_search() in there?
I've been playin around in linux for the last week, and came up with a way to try DVB-S2(x) first, then try DVB-S if no lock.
As long as the SR is "guessed" at about 20% higher than the actual SR, S2 will lock in about 1/2 sec, and 
DVB-2 in about a second, and returns the actual SR.