tbsdtv/linux_media #400: Error lastest pve

nricsi94, 2025-06-17:
Error lastest pve

make -C /root/tbsdriver/media_build/v4l install
make[1]: Entering directory '/root/tbsdriver/media_build/v4l'

Hmm... distro kernel with a non-standard place for module backports detected.
Please always prefer to use vanilla upstream kernel with V4L/DVB
I'll try to remove old/obsolete LUM files from /lib/modules/6.8.12-11-pve//updates/dkms:
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?

Hmm... distro kernel with a non-standard place for module backports detected.
Please always prefer to use vanilla upstream kernel with V4L/DVB
I'll try to remove old/obsolete LUM files from /lib/modules/6.8.12-11-pve//updates/dkms:
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
find: warning: '-name' matches against basenames only, but the given pattern contains a directory separator ('/'), thus the expression will evaluate to false all the time.  Did you mean '-wholename'?
Installing kernel modules under /lib/modules/6.8.12-11-pve/updates/extra/media/:
        ../misc/altera-stapl/: altera-stapl.ko
        pci/tbsecp3/: tbsecp3.ko
        cec/core/: cec.ko
        tuners/: tuner-types.ko tuner-simple.ko si2157.ko
        dvb-frontends/: tas2101.ko si2168.ko
        cec/usb/pulse8/: pulse8-cec.ko
        dvb-core/: dvb-core.ko
/sbin/depmod -a 6.8.12-11-pve
make -C firmware install
make[2]: Entering directory '/root/tbsdriver/media_build/v4l/firmware'
Installing firmwares at /lib/firmware:
make[2]: Leaving directory '/root/tbsdriver/media_build/v4l/firmware'
make[1]: Leaving directory '/root/tbsdriver/media_build/v4l'

crazycat69, 2025-06-18:
i not see any errors, only warnings

closed 2025-06-18 by theone2025-3

nricsi94, 2025-06-18:
Installed old kernel and working and reboot working log:
Hi, I managed to get it working but a little interesting thing is this not a problem for the system?
Another thing is that I use multiswitch, channels work but I don't know if this is a problem or not.
I use this: Goobay 64876 Multiswitch
Could this be a driver problem? Or if it works, I shouldn't bother with it?

root@vpsend:~# dvb-fe-tool --adapter=0 --frontend=0
Device TurboSight TBS 6905 DVB-S/S2  (/dev/dvb/adapter0/frontend0) capabilities:
     CAN_2G_MODULATION
     CAN_FEC_AUTO
     CAN_INVERSION_AUTO
     CAN_MULTISTREAM
     CAN_QPSK
DVB API Version 5.12, Current v5 delivery system: DVBS
Supported delivery systems:
    [DVBS]
     DVBS2
     DSS
Frequency range for the current standard:
From:             950 MHz
To:              2.15 GHz
Symbol rate ranges for the current standard:
From:             100 kBauds
To:              70.0 MBauds
SEC: set voltage to OFF
ERROR    FE_SET_VOLTAGE: Operation not permitted
root@vpsend:~# dvb-fe-tool --adapter=1 --frontend=0
Device TurboSight TBS 6905 DVB-S/S2  (/dev/dvb/adapter1/frontend0) capabilities:
     CAN_2G_MODULATION
     CAN_FEC_AUTO
     CAN_INVERSION_AUTO
     CAN_MULTISTREAM
     CAN_QPSK
DVB API Version 5.12, Current v5 delivery system: DVBS2
Supported delivery systems:
     DVBS
    [DVBS2]
     DSS
Frequency range for the current standard:
From:             950 MHz
To:              2.15 GHz
Symbol rate ranges for the current standard:
From:             100 kBauds
To:              70.0 MBauds
SEC: set voltage to OFF
ERROR    FE_SET_VOLTAGE: Operation not permitted
root@vpsend:~# dvb-fe-tool --adapter=2 --frontend=0
Device TurboSight TBS 6905 DVB-S/S2  (/dev/dvb/adapter2/frontend0) capabilities:
     CAN_2G_MODULATION
     CAN_FEC_AUTO
     CAN_INVERSION_AUTO
     CAN_MULTISTREAM
     CAN_QPSK
DVB API Version 5.12, Current v5 delivery system: DVBS2
Supported delivery systems:
     DVBS
    [DVBS2]
     DSS
Frequency range for the current standard:
From:             950 MHz
To:              2.15 GHz
Symbol rate ranges for the current standard:
From:             100 kBauds
To:              70.0 MBauds
SEC: set voltage to OFF
ERROR    FE_SET_VOLTAGE: Operation not permitted
root@vpsend:~# dvb-fe-tool --adapter=3 --frontend=0
Device TurboSight TBS 6905 DVB-S/S2  (/dev/dvb/adapter3/frontend0) capabilities:
     CAN_2G_MODULATION
     CAN_FEC_AUTO
[... 197 lines cut ...]
[   58.279752] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
[   58.402696] TBSECP3 driver 0000:03:00.0: MAC address 00:22:ab:92:8f:a4
[   58.402707] TBSECP3 driver 0000:03:00.0: DVB: registering adapter 8 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   58.402769] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
[   58.525731] TBSECP3 driver 0000:03:00.0: MAC address 00:22:ab:92:8f:a5
[   58.525741] TBSECP3 driver 0000:03:00.0: DVB: registering adapter 9 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   58.525798] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
[   58.648724] TBSECP3 driver 0000:03:00.0: MAC address 00:22:ab:92:8f:a6
[   58.648735] TBSECP3 driver 0000:03:00.0: DVB: registering adapter 10 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   58.648797] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
[   58.771730] TBSECP3 driver 0000:03:00.0: MAC address 00:22:ab:92:8f:a7
[   58.771741] TBSECP3 driver 0000:03:00.0: DVB: registering adapter 11 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   58.771797] TBSECP3 driver 0000:03:00.0: TurboSight TBS 6909 DVB-S/S2 : PCI 0000:03:00.0, IRQ 45, MMIO 0xfb600000
[   62.024075] TBSECP3 driver 0000:01:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (950000..2150000)
[   62.027920] TBSECP3 driver 0000:01:00.0: DVB: adapter 1 frontend 0 frequency 0 out of range (950000..2150000)
[   62.030015] TBSECP3 driver 0000:01:00.0: DVB: adapter 2 frontend 0 frequency 0 out of range (950000..2150000)
[   62.032674] TBSECP3 driver 0000:01:00.0: DVB: adapter 3 frontend 0 frequency 0 out of range (950000..2150000)
[   62.033531] TBSECP3 driver 0000:03:00.0: DVB: adapter 4 frontend 0 frequency 0 out of range (300000..2350000)
[   62.034415] TBSECP3 driver 0000:03:00.0: DVB: adapter 5 frontend 0 frequency 0 out of range (300000..2350000)
[   62.035585] TBSECP3 driver 0000:03:00.0: DVB: adapter 6 frontend 0 frequency 0 out of range (300000..2350000)
[   62.036460] TBSECP3 driver 0000:03:00.0: DVB: adapter 7 frontend 0 frequency 0 out of range (300000..2350000)
[   62.037294] TBSECP3 driver 0000:03:00.0: DVB: adapter 8 frontend 0 frequency 0 out of range (300000..2350000)
[   62.038206] TBSECP3 driver 0000:03:00.0: DVB: adapter 9 frontend 0 frequency 0 out of range (300000..2350000)
[   62.039049] TBSECP3 driver 0000:03:00.0: DVB: adapter 10 frontend 0 frequency 0 out of range (300000..2350000)
[   62.039941] TBSECP3 driver 0000:03:00.0: DVB: adapter 11 frontend 0 frequency 0 out of range (300000..2350000)